espressif/esp-aws-expresslink-eval #2: ExpressLink C3 doesn't connect

gcohen10, 2022-03-06:
Hello, 

I am trying to connect to AWS IoT using ExpressLink ESP32-C3 Dev Kit but with no success. I have following the steps provided in the README file several times without any luck. 

I have attached a screenshot of the Arduino terminal:
[image]
[image]

dhavalgujar, 2022-03-07:
Hi @gcohen10,
I can see that your Arduino console is set to using "Both NL & CR" as End of Line characters.
This would mean that the EoL is '\r\n' in your case.
As shown in [Section 3.7](https://github.com/espressif/esp-aws-expresslink-eval#37-set-up-host-machine), the EoL will need to be set to 'Line Feed' (The dropdown option is called 'Newline' in case of the Arduino terminal) i.e. '\n'

However, I think that your issue might be different, because if the setup was otherwise correct and only the EoL was incorrect, you would see parsing errors.
Can you check the RX and TX pins (and also interchanging them) when connecting using the USB-to-TTL converter?

If it doesn't work, please send more details of your setup and connections.

HTH.

gcohen10, 2022-03-07:
Hello @dhavalgujar , 

Thanks for your reply,

Before connecting with the Arduino. I connected a USB to TTL converter to the dev kit and tried it out using the guide provided in the README.  I have also tried 'Line Feed' but I am just not getting any responce back from the device. 

When I type AT, I don't get an OK response back. I have also tried to interchanging RX and TX but still not working.

I have followed the exact steps as in the provided guide in the README but the device is just not responding **OK** for a simple **AT** Command.

Looking forward to hearing from you

Best Regards
Gavi

gcohen10, 2022-03-09:
Hello @dhavalgujar 

Any updates regarding the reported issue ? 

Looking forward to hearing from you 

Best Regards
Gavi

dhavalgujar, 2022-03-10:
Hi @gcohen10,
We are unable to reproduce this issue.
Also, since the boards are pre-provisioned and pre-programmed with ExpressLink firmware, it is unlikely for them to be completely dead out-of-the-box.

Can you please send us a diagram and picture of your connections (in case of the USB-to-TTL converter)?

gcohen10, 2022-03-16:
Hello @dhavalgujar , 

Sorry for the delay. 

I have tried to connect the USB-to-TTL converter to RX and TX pins on the the Dev kit and I have also crossed in between them but I am not getting any responce back on the terminal when I type **AT**. 

[image]

I have marked in **RED** and **BLUE** which RX and TX pins I have used to try and communicate with the dev kit using the USB-to-TTL converter. 

Which pins are you using to connect the USB-to-TTL converter to the Dev Kit as I can see 2 RX and TX pins on the Dev Kit. 

Looking forward to hearing from you 

Best Regards
Gavi

pdbayes, 2022-07-31:
I have the very same issue and I have 2 boards exhibiting this behaviour. They are running an older version of firmware but I cannot update it as it always times out. The support for this product is very poor, it would be great if someone could actually help.

dhavalgujar, 2022-07-31:
Hi @gcohen10, I believe this issue has been resolved for you.
Can you please close the issue on GitHub?

@pdbayes my apologies for the issue that you are facing.
I believe you have reported it separately (#7), let us continue to debug your problem there, thanks.