espressif/esp-aws-expresslink-eval #6: Possible to flash this into ESP32-S3 ?

amizer12, 2022-07-22:
Hi,
Is it possible to use this firmware to flash the ESP32-S3 module or other ?
I tried to do this via OTW update process but otw.py throws error:
`_UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa2 in position 0: invalid start byte_`

Which i assume is related to the script itself but not the board.
Can you confirm we can use this on other chips and not only on the dev kit ?

amizer12, 2022-07-22:
So i picked up the ESP32 devkit, erased flash comepltely and flash the expreslink firmware to it and it doesn`t work - serial just spits out constant loop of messages like:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3c0f0020,len:204496
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
load:0x3fc92c00,len:14608
load:0x40380000,len:43016
1162 mmu set 00040000, pos 00040000
load:0x42000020,len:962840
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
ets Jun  8 2016 00:22:57`

not sure if this is the hardware or the firmware is just precisely for the C3 module ?

landretk, 2022-09-21:
No, the binary provided is an update image for ExpressLink devices, and only contains the application OTA image. Flashing a blank device requires more than that(bootloader, partition, keys, etc).

The sales model for ExpressLink as I understand it is that you license a copy of the firmware by buying a module with the firmware on it, it is not freely available to flash on unprovisioned devices. It would not be that useful anyway because you still would need to get keys from Amazon's CA, etc, to provision it.