espressif/esp-aws-expresslink-eval #7: Firmware update

pdbayes, 2022-07-26:
I have 2 of these boards and am trying to update the firmware as AWS IoT Core is not recognising any of the messages sent from the demo Arduino sketch, even though I can see it is connected and sending messages. IoT Core can see it connect and auto subscribes it to some topics. I have tried to update several times and it always times out. I am connecting directly to the board via the micro usb and it is detached from the Arduino, is this correct way to do it?

dhavalgujar, 2022-07-31:
Hi @pdbayes, this is not the correct way to carry out the OTW update.
You can use the microUSB port to power the board and view the ExpressLink console logs, but the only interface for sending and receiving commands is using the RX/TX pins on the board and connecting it to a host machine.

Please refer to [Section 3.6.3](https://github.com/espressif/esp-aws-expresslink-eval#363-with-any-development-host) for making these connections.
Note that you will need to use a USB-TTL converter like the one shown in [User Provided items](https://github.com/espressif/esp-aws-expresslink-eval#33-user-provided-items) in order to connect the ExpressLink board to your computer and issue AT commands / carry out OTW update.

Please post some logs here for the issues that you are facing with sending messages to AWS IoT Core.
Hopefully updating to the latest release will help alleviate those problems.

pdbayes, 2022-07-31:
Thanks for responding, I have been going mad with this 
 OK. So I connect the board to my Arduino. I'm guessing the Arduino needs to be running just the software serial sketch and nothing else? I use my usb to TTL converter and connect to which Rx/tx on the Espressif board? And run the OTW.py on my pc with the port of usb to TTL converter and baud rate as args.  Is this correct.

dhavalgujar, 2022-07-31:
The Arduino sketch has been provided only as a reference to quickly evaluate the ExpressLink DevKit or as a starting point for host-side code if you plan to use an Arduino as a host with ExpressLink.

 You can connect the USB-TTL converter to these RX/TX pins:
[image]
You will not need to set baud rate manually, the otw.py script takes care of the rest.
Please test if AT commands work using a terminal application before running otw.py

pdbayes, 2022-07-31:
OK. Thanks. I understand the Arduino sketch is just a starter example. But I will need the software serial running to communicate won't I?

dhavalgujar, 2022-07-31:
Yes, you will need to use a terminal application configured with the settings from [Section 3.7](https://github.com/espressif/esp-aws-expresslink-eval#37-set-up-host-machine) in order to test if AT commands are working.

The [otw.py](https://github.com/espressif/esp-aws-expresslink-eval/blob/main/tools/otw.py) script uses the pySerial module to communicate with ExpressLink, so you need not configure anything for that.
You will need to install that before running the OTW script, as shown in [Section 9.2](https://github.com/espressif/esp-aws-expresslink-eval#92-carrying-out-an-over-the-wire-otw-upgrade).

pdbayes, 2022-08-01:
Should this work with an Arduino Uno? It only has one UART. Software serial just seems to duplicate pins 1 and 0 which are the same uart the dev board is using

dhavalgujar, 2022-08-01:
Yes, you can use it with an Arduino Uno.
You can use pins 8 and 9 of the Arduino as shown in [Section 8.1](https://github.com/espressif/esp-aws-expresslink-eval#81-debugging-the-arduino-sketch) to get a different software serial.
Of course, for this you will need to connect a USB-TTY dongle to those pins.

pdbayes, 2022-08-01:
I am trying that and I get no response from AT commands

dhavalgujar, 2022-08-01:
Are you trying to use the Arduino Uno as a pass-through serial interface?

Do the AT commands work for you when you try with a USB-TTY dongle and connect to the RX/TX pins of the ExpressLink board directly (with a desktop terminal application)?

pdbayes, 2022-08-01:
I have tried as a passthrough device and also using the sketch provided.
which pins are direct connections, are they the ones in the image above? And should it be disconnected from the Arduino for this to work?

pdbayes, 2022-08-01:
I have a serial connection as pressing the reset button gives be a lot of info ending in this;
I (935) ExpressLink: Thingname is: bd244628-8223-42e9-85d2-fc55383371de

[INFO] [TE] [thirdeye_init:87] Starting ThirdEye

I (945) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo                                                                                                                                                                                                                                                                                                           wn: 0| Intr:0

I (955) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldow                                                                                                                                                                                                                                                                                                           n: 1| Intr:0

I (965) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldow                                                                                                                                                                                                                                                                                                           n: 1| Intr:0

[INFO] [TE] [thirdeye_init:123] Starting MQTT Agent

[INFO] [MQMGR] [bStartMQTTAgent:314] MQTT Agent started successfully

I (985) TIME: Initializing the NTP

[INFO] [APPMAIN] [app_main:125] [INIT] Complete.

[INFO] [APPMAIN] [app_main:130] APP_MAIN complete

I (985) OTA: Current Firmware version 0.9.3

I (995) OTA: Waiting for MQTT agent to be connected.

pdbayes, 2022-08-01:
Swapping RX and TX seems to now give me a response

pdbayes, 2022-08-01:
So, I still can't update the firmware. What needs to be running on the UNO?

dhavalgujar, 2022-08-01:
> So, I still can't update the firmware. What needs to be running on the UNO?

For OTW, please connect to your computer directly using a USB-TTY dongle and run the script, please disconnect it from Arduino during OTW.

> And sent messages to AWS are not showing in the MQQT test window
Can you please post logs from both consoles? (ExpressLink logs and also the console where you issued AT commands)

pdbayes, 2022-08-01:
I will do this later.  I have to do my work for most of the day. It did start the firmware update but failed reading the version number.

pdbayes, 2022-08-02:
It is all working now, thanks for your help. One thing I did notice, when I typed AT+CONF EndPoint=abcdef I got an error, I had to change it to Endpoint. The example sketch is using EndPoint.

mra59, 2022-08-04:
Hi pdbayes, I'm using the same devkit but I cannot connect to AWS: could you tell me what is the firmware revision you have installed into the devkit using the OTW procedure?
Many thanks

pdbayes, 2022-08-04:
It was originally something like 0.9.60. My issue in updating was that you need to cross the Rx and TX of the TTL converter. 
Also in the example code there is possibly an error on the setting the endpoint. It should be Endpoint with no capital P. I provisioned mine manually using the USB ttl converter and it was fine after that.

mra59, 2022-08-04:
Me too discovered the "Endpoint" string and received the DevKit with an old revision.
Using the OTW I flashed that I think is the last revision: v1.0.20.bin.
I have also properly connected the DevKit to my PC (I use Realterm to send commands) using an USB-to-UART converter and the communication is ok.
I get the "OK" answer to the AT Commands that I send but my issue is that it cannot connect to AWS.
I receive the error message "ERR14 4 UNABLE TO CONNECT Failed to securely authenticate with AWS" just 2 o3 seconds after the AT+CONNECT command.
I properly set the device object in AWS (I have already used the IoT Core with other boards) but I don't have any idea about the reason of the ERR14.
Any advice?

pdbayes, 2022-08-04:
Did you copy and upload the certificate from the device and called the thing the same as the board reports as it's thing name?

mra59, 2022-08-04:
Yes: first I downloaded the Certificate from DevKit (AT+CONF? Certificate), then its Thing name (AT+CONF? ThingName).
Then I created in my AWS IoT Core account the object with the ThingName and I created a Policy allowing all operations from any arn.
Then I attaced the certificate (activated) to the policy and I obtained the endpoint from AWS (1234567890.iot.eu-west-1.amazonaws.com)
This endpoint was transferred to the Devkit with the AT+CONF Endpoint=1234567890.iot.eu-west-1.amazonaws.com.
Now I'm ready to connect but I always receive the "ERR14 4 ......"  message.
I checked the WiFi access point and I can see the "espressif" connected with its IP address and MAC.
The connection time is only 6 seconds!!!!!!!

dhavalgujar, 2022-08-04:
Hi @mra59, can you post the logs that you are seeing on the ExpressLink console i.e. UART1 from the microUSB port on your original issue #8?

mra59, 2022-08-04:
Hi dhavalgujar, I tried two ways: the first is connecting the DevKit to my PC through a USB-to-UART converter.
I connected Rx and tx lines to the Arduino pins 0 and 1 so that I can manually send commands using the RealTerm software.
In this way I receive the famous "ERR14 4 ...." errore message.
The second way I tried was the use of the "Start_Quick_Connect" software: in this case I connected my PC directly to the Micro USB port of the Devkit.
Before running the "Start_Quick_Connect" I setup the serial port, the SSID and the Passphrase in the "Config.txt" file used by the "Start_Quick_Connect" software.
Despite this the "Start_Quick_Connect" ask me to insert all the parameters needed to connect: SSID, Passphrase, ThingName and Endpoint: I answered to these questions manually using the RealTerm connected with the USB-to_UART Converter to the Arduino Rx and tx pins.
After finishing  I get a little different error that is "Thing doesn't exist" (see the attached report).
To resume: sending manually AT+CONNECT I get the message "ERR14 4 UNABLE TO CONNECT Failed to securely authenticate with AWS", using the QuickConnect software I get the "Status Code 400, (Thing doesn't exist)"

[Connection_Error_StatusCode_400.pdf](https://github.com/espressif/esp-aws-expresslink-eval/files/9261834/Connection_Error_StatusCode_400.pdf)

mra59, 2022-08-04:
I attach again the last part of the log of the "Start Quick Connect" software because the previous one was not complete.
[Connection_Error_StatusCode_400.pdf](https://github.com/espressif/esp-aws-expresslink-eval/files/9261885/Connection_Error_StatusCode_400.pdf)

mra59, 2022-08-04:
Maybe the reason could be that my AWS account is registered in Ireland (eu-west-1)  and the Devkit object/certificate is valid only in other regions (pacific ap-east-1 or ap-south-1) ?

dhavalgujar, 2022-08-04:
@mra59  Thank you for sending these logs.
Can you also send the logs from the serial port that you get when you connect the ExpressLink to your computer via microUSB?
When you reset the board you will see the bootup log on this port.
That will help us investigate the issue that you are facing.

There is no region restriction for using ExpressLink devkits.

mra59, 2022-08-04:
This is the log at DevKit startup.
[ESP32_C3_MINI-1-AWS_ExpressLink_StartupMessage_20220804.txt](https://github.com/espressif/esp-aws-expresslink-eval/files/9261973/ESP32_C3_MINI-1-AWS_ExpressLink_StartupMessage_20220804.txt)

dhavalgujar, 2022-08-04:
Bootup looks okay, you've successfully updated to v1.0.20 of the firmware.

Can you see what logs you get on this console, when you try to use the `AT+CONNECT` command?

The log will contain your SSID and Passphrase, please redact them before posting the log here.

mra59, 2022-08-04:
Attached the log of the 3 commands.
The DevKit is properly connected to my WiFi network (I'm using my smartphone as Access Point in this moment) because I see that it's registered as "espressif" with its IP address and its MAC number.
[ERR14_Log_20220804.txt](https://github.com/espressif/esp-aws-expresslink-eval/files/9262048/ERR14_Log_20220804.txt)

dhavalgujar, 2022-08-04:
When you run these commands, you'll also see corresponding debug logs on the console where you could see the bootup log.
Basically you need to simultaneously open two consoles, one where you will give the AT command and the other where you can see what the module is doing.

We should be able to better diagnose your issue if we can look at those logs.
Does this make sense?

mra59, 2022-08-04:
Nice idea.
I opened two istances of the RealTerm, one to manually send the AT+CONNECT command and the second istance connected to the DevKit USB port.
Attached the Log of the USB port.
[ESP32-C3_USB_Port_Connection_Log.txt](https://github.com/espressif/esp-aws-expresslink-eval/files/9262104/ESP32-C3_USB_Port_Connection_Log.txt)

dhavalgujar, 2022-08-04:
Your endpoint seems to be incorrect, can you try setting it again using the `AT+CONF Endpoint=<endpoint>` command?

mra59, 2022-08-04:
[ESP32-C3_USB_Port_SetEndPoint_TryConnect_Log.txt](https://github.com/espressif/esp-aws-expresslink-eval/files/9262133/ESP32-C3_USB_Port_SetEndPoint_TryConnect_Log.txt)

mra59, 2022-08-04:
I attach a screenshot of the object and its endpoint I created in my AWS account.
[image]

dhavalgujar, 2022-08-04:
You seem to be confusing the ARN of your Thing, i.e. `arn:aws:iot:eu-west-1:880869895289:thing/724bc322-581a-4ed7-8d1d-ca3aa31f88e5` with the Endpoint of your AWS account. :)
The endpoint would resemble something like: abcd1234-ats.iot.eu-west-1.amazonaws.com

You can find yours on the [Settings](https://console.aws.amazon.com/iot/home#/settings) page.

mra59, 2022-08-04:
I set AT+CONF Endpoint=agg23zwk76whx-ats.iot.eu-west-1.amazonaws.com and now it works!!!
I'm very grateful to you but I'm very sure that I tried also this Endpoint value during all the tests I performed to let the Devkit working.
Anyway I thank you very much also for the advice to use the USB port as logger of what happen inside.
Many thanks

dhavalgujar, 2022-08-04:
Glad it worked out for you, @mra59!

---
@pdbayes, the sketch will be updated shortly with a few fixes to the serial read logic and like you mentioned, the updated AT commands.
Since the issue you reported originally is resolved, I will close this one.
Feel free to open a new issue in case you run into any other problems.

closed 2022-08-04 by dhavalgujar

geb212, 2023-06-01:
@dhavalgujar 

[image]

Hi, for me these pin doesn't work, I use the pin gnd, tx and x of the ack con the board, ait's worked, but I can't uate the firmware, what I have to do?
Thanks