espressif/esp-aws-expresslink-eval #10: Extra characters appended/repeated for CONFMODE command

andysargent-cp, 2022-09-16:
Command from processor:
AT+CONFMODE WPKCBXDGW8I09HI5W326J4QKMBJ9

Results in an advertisement that is not the same.  In addition, the POP mode is enabled in the app when this happens.

 (Note the 'WPKCBXDGW8I09HI5W326J4QKMBJ9J4QKMBJ9' has and extra 'J4QKMBJ9' repeated on the end!)

I (3303) SendToHost: writing : OK 0 DISCONNECTED

W (3333) BTDM_INIT: esp_bt_mem_release not implemented, return OK
I (3333) wifi_prov_scheme_ble: BT memory released
W (3333) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (3333) BTDM_INIT: BT controller compile version [d913766]
I (3343) BTDM_INIT: Bluetooth MAC: 60:55:f9:cb:01:4a

I (3353) protocomm_nimble: BLE Host Task Started
I (3353) NimBLE: GAP procedure initiated: stop advertising.

I (3363) NimBLE: GAP procedure initiated: advertise;
I (3363) NimBLE: disc_mode=2
I (3373) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (3383) NimBLE:

I (3383) wifi_prov_mgr: Provisioning started with service name : WPKCBXDGW8I09HI5W326J4QKMBJ9J4QKMBJ9"
I (3393) CONFMODE: Provisioning started
I (3393) SendToHost: writing : OK CONFMODE ENABLED

I (3433) SendToHost: writing : OK 2 0 STARTUP

I (4453) SendToHost: writing : OK 3 0 CONLOST

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (5473) SendToHost: writing : OK

Logic analyzer indicates that only 'AT+CONFMODE WPKCBXDGW8I09HI5W326J4QKMBJ9' is sent.

[image]

This issue seems to happen after resetting from a connected state after setting SSID to "" and resetting.

andysargent-cp, 2022-09-16:
Forgot to specify Current Firmware version 1.0.21

avsheth, 2022-09-22:
Hi @andysargent-cp 
I did a quick check and was successfully able to set the service name without PoP. Below is the log of the same.

As of now STARTUP event is only sent during the boot and when EL comes out of the sleep mode. Hence I am unsure how it made its way in your log in the midst of a session. 

Could you share :
1. If you've made any changes in the Arduino sketch
2. If host isn't Arduino, could you share list of all commands before CONFMODE?
3. And complete runtime log of EL module

```
W (3080) BTDM_INIT: esp_bt_mem_release not implemented, return OK
I (3080) wifi_prov_scheme_ble: BT memory released
W (3080) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (3080) BTDM_INIT: BT controller compile version [3a49744]
I (3090) BTDM_INIT: Bluetooth MAC: 84:f7:03:3f:20:7a

I (3100) protocomm_nimble: BLE Host Task Started
I (3100) NimBLE: GAP procedure initiated: stop advertising.

I (3110) NimBLE: GAP procedure initiated: advertise;
I (3110) NimBLE: disc_mode=2
I (3120) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (3130) NimBLE:

I (3130) wifi_prov_mgr: Provisioning started with service name : WPKCBXDGW8I09HI5W326J4QKMBJ9
I (3140) CONFMODE: Provisioning started
I (3140) SendToHost: writing : OK CONFMODE ENABLED

I (37580) NimBLE: GAP procedure initiated: advertise;
I (37590) NimBLE: disc_mode=2
I (37590) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (37590) NimBLE:

[INFO] [WIFI] [WIFI_EventHandler:184] WiFi Event Handler
[INFO] [WIFI] [WIFI_EventHandler:196] Wifi Scan Done
I (79270) CONFMODE: Received Wi-Fi credentials
        SSID     : xxxxxx
        Password : 000000
I (82690) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (83360) wifi:state: init -> auth (b0)
I (83360) wifi:state: auth -> assoc (0)
I (83370) wifi:state: assoc -> run (10)
I (83410) wifi:connected with note10_lite, aid = 3, channel 11, BW20, bssid = b6:42:ff:ff:8d:06
I (83410) wifi:security: WPA2-PSK, phy: bgn, rssi: -30
I (83420) wifi:pm start, type: 1

I (83420) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (83430) wifi:BcnInt:102400, DTIM:2
[INFO] [WIFI] [WIFI_EventHandler:184] WiFi Event Handler
[INFO] [WIFI] [WIFI_EventHandler:187] Station connected
W (83440) wifi:<ba-add>idx:0 (ifx:0, b6:42:ff:ff:8d:06), tid:0, ssn:0, winSize:64
I (84010) esp_netif_handlers: sta ip: 192.168.21.85, mask: 255.255.255.0, gw: 192.168.21.249
[INFO] [WIFI] [WIFI_EventHandler:216] Station received an IP address.
I (84010) wifi_prov_mgr: STA Got IP
I (84010) CONFMODE: Provisioning successful
I (87750) NimBLE: GAP procedure initiated: stop advertising.

I (87760) NimBLE: GAP procedure initiated: stop advertising.

I (87760) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

E (87790) protocomm_nimble: Error setting advertisement data; rc = 30
W (87810) Timer: Timer not stopped
W (87820) Timer: Timer not stopped
I (87850) wifi_prov_mgr: Provisioning stopped
W (87850) BTDM_INIT: esp_bt_mem_release not implemented, return OK
I (87850) wifi_prov_scheme_ble: BTDM memory released
I (87870) SendToHost: writing : OK 7 0 CONFMODE

[INFO] [CONNECT] [elCommand_connect:218] ExpressLink CONNECT requested
[INFO] [CONF] [elCommand_confQuery:169] Reading Endpoint[0]
[INFO] [ENDPOINT] [endpointReader:49] Read Endpoint=espressif.expresslink.iot.us-west-2.on.aws from flash
[INFO] [CONF] [elCommand_confQuery:178] -----> espressif.expresslink.iot.us-west-2.on.aws
[INFO] [CONNECT] [mqttConnect:169] Creating TLS connection to espressif.expresslink.iot.us-west-2.on.aws.
[INFO] [MQMGR] [expresslink_tlsConnect:82] Successfully obtained the ds context
[INFO] [MQMGR] [expresslink_tlsConnect:83] The value of rsa length is 2048
[INFO] [MQMGR] [expresslink_tlsConnect:84] The value of efuse key id is 0
[INFO] [CONNECT] [mqttConnect:182] Created TLS connection to IoT Core
[INFO] [CONNECT] [mqttConnect:188] Creating an MQTT connection to espressif.expresslink.iot.us-west-2.on.aws.
[INFO] [CONF] [elCommand_confQuery:169] Reading CustomName[0]
[INFO] [CUSTOMNAME] [customNameReader:56] Read CustomName= from flash
[INFO] [CONF] [elCommand_confQuery:178] ----->
```

andysargent-cp, 2022-09-26:
Here is the full log.

-Andy

-- 

-Andy Sargent
Cardinal Peak

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to eve                                                                         nt queue
I (257579) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to eve                                                                         nt queue
I (258589) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to eve                                                                         nt queue
I (259599) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (260609) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (261619) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (262629) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (263639) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (264649) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (265659) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (266669) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (267679) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (268689) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (269699) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (270709) SendToHost: writing : OK

[INFO] [CONF] [elCommand_conf:148] Writing SSID[0]
[INFO] [WIFI] [ExpressLink_ConfigHandler_WifiSsidWrite:79] Wrote SSID= to flash
I (270739) SendToHost: writing : OK

I (270759) SendToHost: writing : OK

[INFO] [WIFI] [WIFI_EventHandler:184] WiFi Event Handler
[INFO] [WIFI] [WIFI_EventHandler:209] WIFI_EVENT event_id: 3
I (270759) wifi:flush txq
I (270759) wifi:stop sw txq
I (270759) wifi:lmac stop hw txq
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806c8
SPIWP:0xee
mode:DIO, clock div:1
Valid secure boot key blocks: 0
secure boot verification succeeded
[... 715 lines cut ...]
I (12883) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (13903) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (14923) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (15943) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (16963) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (17983) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (19003) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (20023) SendToHost: writing : OK

[ERROR] [EVENTS] [ExpressLink_Events_Pop:147] Failed to pop event:UNKNOWN to event queue
I (21043) Se

andysargent-cp, 2022-09-26:
[ExpressLink Bug Full Log.txt](https://github.com/espressif/esp-aws-expresslink-eval/files/9646362/ExpressLink.Bug.Full.Log.txt)