espressif/esp-aws-expresslink-eval #11: Can't download firmware on module with ESP-IDF tool

davimarriel, 2022-09-20:
When I try to download new firmware on the board using the ESP-IDF tool via USB I get the bellow image on my console

[image]

How can I solve it?

landretk, 2022-09-21:
The ExpressLink module is read-only and has secure boot enabled, so it can only be updated through OTA or with the OTW.py tool that flashes through the AT command interface(not the USB), see here for directions for using the .bin with otw.py https://github.com/espressif/esp-aws-expresslink-eval#92-carrying-out-an-over-the-wire-otw-upgrade

hughwphd, 2023-05-04:
> The ExpressLink module is read-only and has secure boot enabled, so it can only be updated through OTA or with the OTW.py tool that flashes through the AT command interface(not the USB), see here [...]

I ran the OTW but got the following errors:

[INFO] [CONF] [elCommand_confQuery:169] Reading Version04800[0]
[INFO] [ABOUT] [versionReader:70] Read Version
[INFO] [CONF] [elCommand_confQuery:178] -----> 1.0.20
I (5633265) SendToHost: writing : OK 1.0.20

[INFO] [OTW] [expresslink_otw:118] Total size 1314816, Block size 204800
[INFO] [OTW] [expresslink_otw:124] Starting OTW
[INFO] [OTW] [expresslink_otw:127] Boot Partition at 0x00020000 of 001D0000
[INFO] [OTW] [expresslink_otw:129] Running Partition at 0x00020000 of 001D0000
[INFO] [OTW] [expresslink_otw:133] Update Partition at 0x001F0000 of 001D0000
[INFO] [OTW] [expresslink_otw:142] Scratch buffer allocation error
I (5633315) SendToHost: writing : ERR1 OVERFLOW

dhavalgujar, 2023-10-28:
Thanks @landretk!
You are absolutely right, closing this for now. :)

closed 2023-10-28 by dhavalgujar