espressif/esp-aws-expresslink-eval #15: ExpressLink SLEEP / Low Power Modes

attila-tokes, 2023-03-03:
Hi! 🙂

I'm currently evaluating the ESP32-C3-MINI-1-N4-A module using the eval board, and I would like to know what Sleep / Low Power modes are implemented in the module.

The AWS ExpressLink specification ([link](https://docs.aws.amazon.com/iot-expresslink/v1.1/programmersguide/elpg-commands.html#elpg-power-control)) defines a `SLEEP` command which supposed to be used to bring the ExpressLink module into sleep low power modes. The modules can implement multiple SLEEP modes, from `0` --> `9` *(lowest power)*. Additionally a timeout can be specified.

I wanted to ask / confirm which of these modes are implemented in the ESP32-C3-MINI-1-N4-A module?

Actually, I tried out the `SLEEP` commands, with and without timeout, multiple modes. The module always acknowledges the commands with `OK`.

However, I did not observed any significant reduction in the current consumption of the module. It remains about 20 mA, which is about the same as in normal idle conditions. At transmit the current consumption is around 80mA. 

The current consumption was measured via the J7 header, using a Nordic Power Profiler Kit II.

I would kind of expected the SLEEP modes to bring the ESP32-C3 into a light sleep *(~2mA)* or deep sleep *(~500uA)* modes.

Thanks!

attila-tokes, 2023-03-04:
Update:

According to the ExpressLink specification the command the modules should implement `SLEEP[#mode]` commands in a backwards compatible manner, meaning that when a sleep mode *(ex. `9`)* is not supported the module should enter the next lowest implemented mode *(ex `3`)*.

It turns out the actual sleep mode used should be included in the response to the `SLEEP[#mode]` command, as `OK {#actual-mode}`. The `ESP32-C3-MINI-1-N4-A` apparently always response with `OK 0`, no matter what the requested sleep mode is. Ex:
```
> AT+SLEEP5
< OK 0
```

From this I assume only Sleep Mode `#0` is implemented in the module. Based on the current consumption, looks like this mode does not actually brings the ESP32-C3 into any kind of sleep / low power mode.

Is this correct? If yes, do you have any plans to implement lower power sleep modes?

Firmware version: 2.4.1

dhavalgujar, 2023-10-28:
Hi @attila-tokes,
Light sleep and deep sleep modes are available now, as a part of the [v2.5.0](https://github.com/espressif/esp-aws-expresslink-eval/releases/tag/v2.5.0) release.

closed 2023-10-28 by dhavalgujar